ros-perception/opencv_apps #2: optical flow in opencv_apps

vrabaud, 2016-05-07:
Hi! First of all I'm quite new to ROS so I'm not sure whether this is a silly question.
copied from https://github.com/ros-perception/vision_opencv/issues/67

What role do the `opencv_apps` play in this repository? Are those meant just as code examples and not meant to be used within 'proper' applications? 

In short: is it advisable to use `opencv_apps/Flow.msg` and `fback_flow_nodelet` in my application or should I roll my own? The fact that `fback_flow_nodelet` unconditionally draws the flow visualization suggests that it's really meant more as an example application, rather than a general nodelet for optical flow.

---

**If** `opencv_apps` are meant to be useful apart from serving as a code example I would suggest a few changes (and gladly provide pull requests) to the optical flow part.

In particular I think that the definition of the flow messages should be different. In `Flow.msg` the flow vector is called `velocity`, but what `fback_flow_nodelet` and `lk_flow_nodelet` really compute are the motions between two frames in pixels irrespective of the time passed between the two frames. So either the nodelets need to scale the movement by the time between the frames, so that the `velocity` field in `Flow.msg` get a proper unit (px/s e.g.), or `velocity` needs to be renamed to `displacement` and a `time` field needs to be added to `Flow.msg`.

PS: great repo! :+1: :smile:

fujimo-t, 2019-03-15:
I created small [ROS packages for DIS Flow](https://github.com/ActiveIntelligentSystemsLab/dis_flow_ros) using OpenCV implementation and consider to create pull request to opencv_apps.

But FlowArray.msg is not suitable for dense flow as pointed in https://github.com/ros-perception/vision_opencv/issues/67#issuecomment-127205720 .
Is it OK to add new msg for dense flow if I send pull request?

[My current implementation](https://github.com/ActiveIntelligentSystemsLab/dis_flow_ros/blob/master/dense_flow_msg/msg/DenseFlow.msg) is just wrapping sensor_msgs/Image like stereo_msgs/DisparityImage.
The reason of this is that Image has enough information to handle data array, and can be accessed easily by cv_bridge.

k-okada, 2019-03-18:
Patches are always welcome, so feel free to create new PR.
- I prefer not to add new messages, especially just for one node, because we have to create viewer, test code and so on just for that node 
- I like the comparison of algorithms, so if your new node outputs the same message as other existing nodes, then we're easy to compare the outputs
  - so if you can implement multiple dense optical flow algorithm, that's would be good
  - or enable to output new message for all existing optical flow algorithm

fujimo-t, 2019-03-18:
I see.

Then I will create PR using existing message `opencv_apps/Flow.msg`.
It is not convinience for dense flow but also can be used.

Later I try to implement other dense algorithms or change to new message if I have time.