ros-perception/opencv_apps #10: Proposal of separating ROS messages into an another package (maybe cv_msgs?) like pcl_msgs

wkentaro, 2016-05-31:
There are many reusable ROS messages in this package.
Why not separating them into an another package like [pcl_msgs](https://github.com/ros-perception/pcl_msgs)?

k-okada, 2016-06-01:
I tried to discussion this topic at https://github.com/ros-perception/vision_opencv/pull/56#issue-53170691
> this package contains a lot of new message, show we create opencv_msgs that holds opencv basic data structure ? or ask other packages (geometry_msgs or others ) to add new messages?

Basic question is 
- current msg try to support opencv types as ros messages, and would like to know if this is the right direction
- if this is ok, we need to check if current implementation covers all opencv data types

wkentaro, 2016-06-01:
> current msg try to support opencv types as ros messages, and would like to know if this is the right direction

IMO, the types in opencv are mostly well refined and creating message for that is reasonable thing, because it seems that OpenCV is more famous than ROS.
But I also think we can reuse messages in geometry_msgs for some messages in this package, for example 
- [geometry_msgs/Point.msg](https://github.com/ros/common_msgs/blob/jade-devel/geometry_msgs/msg/Point.msg) for [opencv_apps/Point2D.msg](https://github.com/ros-perception/opencv_apps/blob/indigo/msg/Point2D.msg) with `z=0`.
- [geometry_msgs/Polygon.msg](https://github.com/ros/common_msgs/blob/jade-devel/geometry_msgs/msg/Polygon.msg) for [opencv_apps/Contour.msg](https://github.com/ros-perception/opencv_apps/blob/indigo/msg/Contour.msg) with `points[:].z=0`.

> if this is ok, we need to check if current implementation covers all opencv data types

Maybe we can add new messages when we need. We can discuss about adding a new message at that time.