ros-perception/opencv_apps #50: c&p mistake in fback_flow documentation on ROS wiki?

HWiese1980, 2016-12-12:
I think there's a copy&paste mistake in the documentation of the fback_flow node in the ROS wiki... I understand that fback_flow is not used for face recognition, so the framed part doesn't make much sense:
[image]

k-okada, 2016-12-12:
Yes, you're correct. Can you fix the wiki?

HWiese1980, 2016-12-12:
I could if I had an account and the time to figure out what actually needed to be written there. It's not the only place where this mistake appears. There are several other occurrences further down.

k-okada, 2016-12-13:
Thanks, you can get accout from here ->
http://wiki.ros.org/action/newaccount/Get%20Involved?action=newaccount

HWiese1980, 2017-01-06:
That helps with the account. Though unfortunately I still don't have the time to take care of this...

k-okada, 2017-01-10:
ok, fixed. please check  http://wiki.ros.org/opencv_apps

closed 2019-03-15 by k-okada

hh129sss5, 2022-03-28:
> ok, fixed. please check  http://wiki.ros.org/opencv_apps

Denied only for overview

HWiese1980, 2022-03-29:
Completely forgot about this one. Thanks, @k-okada!