ros-perception/opencv_apps #51: Launch file crashes on giving a real image topic.

parhartanvir, 2017-03-03:
Whenever I run the following command
`roslaunch opencv_apps lk_flow.launch image:=one_image_raw`

It crashes with the following error

`[lk_flow-1] process has died [pid 6645, exit code -11, cmd /home/frcvision/catkin_ws/devel/lib/opencv_apps/lk_flow image:=one_image_raw __name:=lk_flow __log:=/home/frcvision/.ros/log/c914a1be-ffac-11e6-88c1-d8cb8ac73819/lk_flow-1.log].
log file: /home/frcvision/.ros/log/c914a1be-ffac-11e6-88c1-d8cb8ac73819/lk_flow-1*.log`

The file **lk_flow-1*.log** is always empty.

Thanks,
Tanvir

iory, 2017-03-30:
If you installed ros-indigo-opencv3, please try following steps.
```
mkdir -p ~/ws_opencv_apps/src
cd ~/ws_opencv_apps/src
cd ~/ws_opencv_apps
catkin init
cd ~/ws_opencv_apps/src
wstool init
wstool set --git opencv_apps https://github.com/ros-perception/opencv_apps.git -y
rosinstall_generator image_pipeline --upstream >> .rosinstall.opencv3; fi # need to recompile image_proc
rosinstall_generator compressed_image_transport --upstream >> .rosinstall.opencv3; fi # need to recompile compressed_image_transport
rosinstall_generator vision_opencv --upstream >> .rosinstall.opencv3; fi # need to recompile visoin_opencv
wstool merge .rosinstall.opencv3; fi # need to recompile visoin_opencv
wstool up

source /opt/ros/indigo/setup.bash
catkin b
source ~/ws_opencv_apps/devel/setup.bash
```

k-okada, 2017-03-31:
@iory why you assume @parhartanvir is using opencv3
@parhartanvir running with opencv3 on indigo is super experimental settings, we do not recommend to use that, even if works for opencv_apps, it will have side-effects to other repository
@parhartanvir please provide OS version, ROS version when you run your application, 
```
roslaunch --screen opencv_apps lk_flow.launch image:=one_image_raw
```
will show more information and also `lk_flow-1*.log` is empty, please check other files in `/home/frcvision/.ros/log/c914a1be-ffac-11e6-88c1-d8cb8ac73819/`

iory, 2017-03-31:
> why you assume @parhartanvir is using opencv3

I have seen a similar error, and I suspected it was the effect of opencv3.

hh129sss5, 2022-03-28:
> Closed 

Overview closed