ros-perception/opencv_apps #55: -1 for subscribing camera_info which is unnecessary in actual

wkentaro, 2017-03-20:
I'm not sure why some nodes in opencv_apps does subscribe camera_info even though it is unnecessary.

For example:

### AddingImages

It subscribes camera_info by default with `use_camera_info:=true`:
https://github.com/ros-perception/opencv_apps/blob/indigo/src/nodelet/adding_images_nodelet.cpp#L116

And frame_id is passed to the `do_work` function:
https://github.com/ros-perception/opencv_apps/blob/indigo/src/nodelet/adding_images_nodelet.cpp#L103

But the `input_frame_from_msg` is not used in actual:
https://github.com/ros-perception/opencv_apps/blob/indigo/src/nodelet/adding_images_nodelet.cpp#L164

I wonder there are any cases where the output frame_id is different from both two image inputs.
Cases like below are assumed by default?

- `camera/rgb/camera_info`, `fraem_id=camera_rgb_optical_frame`
- `image_publisher/output1`, `frame_id=dummy_camera_1`
- `image_publisher/output2`, `frame_id=dummy_camera_2`
- `adding_images/image camera_info:=camera/rgb/camera_info image1:=image_publisher/output1 image2:=image_publisher/output2`, `fraem_id=camera_rgb_optical_frame`

k-okada, 2017-03-20:
It is just a templates, to avoid discussion for each node, for same reason
we basically have node if that is a part of opencv example.

Any problem with use_camera_info:=false? If you get used to other
opencv_apps nodes, then you can easily imagine that the node has this
option.

Or if you think default value of use_camera_info should be false, please
send PR.
https://github.com/ros-perception/opencv_apps/blob/indigo/cfg/AddingImages.cfg#L9

wkentaro, 2017-03-20:
> It is just a templates, to avoid discussion for each node, for same reason
we basically have node if that is a part of opencv example.

I mean the template is not good, and wonder what is the merit of the templating. 

> Any problem with use_camera_info:=false? If you get used to other
opencv_apps nodes, then you can easily imagine that the node has this
option.

The package opencv_apps is relatively new, so I mean the design should be considered properly, seeing prior works.
Of course, the `use_camera_info:=false` works, but I dislike to do things that is uncessary.

wkentaro, 2017-05-10:
Fixed via https://github.com/ros-perception/opencv_apps/pull/58

closed 2017-05-10 by wkentaro

hh129sss5, 2022-03-28:
> Open access on this approved M overview approved  it 

Granted