ros-perception/opencv_apps #65: face detect freezes after face is detected.

anfederman, 2017-05-30:
I am running opencv_apps face_detect on an AMD 64 running Ubuntu 16.04 and Kinetic. Immediately after detecting a face, the terminal comes back with "init_done" the image of the face circle changes from purple to grey and then the whole image grays out. The face detection topics are still running, but no data is being echoed. As far as I can tell the process is still running, but kill -HUP or control-c terminates. camera data is still being transmitted.

I am not seeing any error messages.  Any place in logs where I can look to trouble-shoot?

-----
SUMMARY
========

PARAMETERS
 * /face_detection/debug_view: True
 * /face_detection/eyes_cascade_name: /usr/share/opencv...
 * /face_detection/face_cascade_name: /usr/share/opencv...
 * /face_detection/use_camera_info: False
 * /rosdistro: kinetic
 * /rosversion: 1.12.7

NODES
  /
    face_detection (opencv_apps/face_detection)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[face_detection-1]: started with pid [3563]
init done

------

anfederman, 2017-05-31:
Additional information, after freeze, and init done message, all nodes still appear in rostopic list, however none are recieving any data, even the camera nodes. 

No error messages in logs.

k-okada, 2017-06-20:
thank you for reports, please try if #67 solves your problem

closed 2017-07-14 by k-okada