ros-perception/opencv_apps #72: Regression in Kinetic re OpenCV 3.3

tfoote, 2017-11-15:
http://build.ros.org/view/Kbin_uX64/job/Kbin_uX64__opencv_apps__ubuntu_xenial_amd64__binary/45/console
```
00:10:57.242 /usr/lib/ccache/x86_64-linux-gnu-g++   -DROSCONSOLE_BACKEND_LOG4CXX -DROS_PACKAGE_NAME=\"opencv_apps\" -Dopencv_apps_EXPORTS -I/tmp/binarydeb/ros-kinetic-opencv-apps-1.12.0/obj-x86_64-linux-gnu/devel/include -I/tmp/binarydeb/ros-kinetic-opencv-apps-1.12.0/include -I/opt/ros/kinetic/include -isystem /opt/ros/kinetic/include/opencv-3.3.1 -isystem /opt/ros/kinetic/include/opencv-3.3.1/opencv -I/opt/ros/kinetic/share/xmlrpcpp/cmake/../../../include/xmlrpcpp  -g -O2 -fstack-protector-strong -Wformat -Werror=format-security -DNDEBUG -Wdate-time -D_FORTIFY_SOURCE=2  -fPIC   -o CMakeFiles/opencv_apps.dir/src/nodelet/face_recognition_nodelet.cpp.o -c /tmp/binarydeb/ros-kinetic-opencv-apps-1.12.0/src/nodelet/face_recognition_nodelet.cpp
00:11:06.299 /tmp/binarydeb/ros-kinetic-opencv-apps-1.12.0/src/nodelet/face_recognition_nodelet.cpp: In member function ‘void face_recognition::FaceRecognitionNodelet::configCallback(face_recognition::FaceRecognitionNodelet::Config&, uint32_t)’:
00:11:06.299 /tmp/binarydeb/ros-kinetic-opencv-apps-1.12.0/src/nodelet/face_recognition_nodelet.cpp:466:22: error: ‘createEigenFaceRecognizer’ is not a member of ‘face’
00:11:06.299              model_ = face::createEigenFaceRecognizer(config.model_num_components,
00:11:06.299                       ^
00:11:06.299 /tmp/binarydeb/ros-kinetic-opencv-apps-1.12.0/src/nodelet/face_recognition_nodelet.cpp:469:22: error: ‘createFisherFaceRecognizer’ is not a member of ‘face’
00:11:06.299              model_ = face::createFisherFaceRecognizer(config.model_num_components,
00:11:06.299                       ^
00:11:06.299 /tmp/binarydeb/ros-kinetic-opencv-apps-1.12.0/src/nodelet/face_recognition_nodelet.cpp:472:22: error: ‘createLBPHFaceRecognizer’ is not a member of ‘face’
00:11:06.299              model_ = face::createLBPHFaceRecognizer(config.lbph_radius,
00:11:06.299                       ^
00:11:06.302 CMakeFiles/opencv_apps.dir/build.make:473: recipe for target 'CMakeFiles/opencv_apps.dir/src/nodelet/face_recognition_nodelet.cpp.o' failed
```

Re: ros-infrastructure/rep#140 @vrabaud @mikaelarguedas FYI

k-okada, 2017-11-15:
Thanks 

We have just fixed this issue yesterday, and will be released after merging several existing PR, should be within this week.
https://github.com/ros-perception/opencv_apps/pull/71

BTW, we have huge regressions on pluginlib, white all nodelet crashes when these is a package forget to install nodelet xml, like gmapping . Hope you have time to look into this.
https://github.com/ros/pluginlib/pull/85

mikaelarguedas, 2017-11-15:
@k-okada Yes I'm looking into it and will review this and the corresponding nodelet PR shortly

mikaelarguedas, 2017-11-18:
> BTW, we have huge regressions on pluginlib, white all nodelet crashes when these is a package forget to install nodelet xml, like gmapping . Hope you have time to look into this.
ros/pluginlib#85

@k-okada Fix has been merged and will be available in the main repo after the next sync of kinetic and lunar.

closed 2019-03-15 by k-okada