ros-perception/opencv_apps #76: Threshold being reset to 50 for hough_lines

hydroflax, 2018-05-06:
Hi,

The threshold for some reason when using roslaunch params keeps on being reset to 50 whenever a threshold below 50 is used. The bug can be seen by setting the threshold to 25 in the .launch file provided and adding a `NODELET_DEBUG(%d, threshold_);` on line 217 of hough_lines_nodelet.cpp

hydroflax, 2018-05-12:
Hi,

I managed to increase the threshold by decreasing the size of theta. Was just thinking, wouldn't it be an idea to have the slider in the debug_view go from 50-150 if that is the range?

danieljimeneznz, 2019-01-01:
closing issue since there is a workaround.

closed 2019-01-01 by danieljimeneznz

k-okada, 2019-01-08:
@danieljimeneznz what is the "workaround" ? 
from https://github.com/opencv/opencv/blob/master/modules/imgproc/src/hough.cpp#L174-L188, I think range 50-150 is not correct, it could be 1 to width*height.

hh129sss5, 2022-03-28:
Denied should money on this is only for overview and the other matter should be only 1