ros-perception/opencv_apps #86: Debug view: Publish image instead opening OpenCV window

fujimo-t, 2019-03-15:
Why does opencv_apps use OpenCV's window and not use ROS topic to output debug view?

I think using ROS topic better than OpenCV's window:

- The topic can be seen from other machines by ROS's network system
- Can be seen with output of other node by existing tools like rqt or RViz

k-okada, 2019-03-18:
if you run the node with `debug_view:=false` then you can run the node without OpenCV window.

https://github.com/ros-perception/opencv_apps/blob/indigo/launch/adding_images.launch#L8

fujimo-t, 2019-03-19:
> if you run the node with debug_view:=false then you can run the node without OpenCV window.
Yes, I know it.

What I want to say is not that the window is annoying.
Just I think ROS topic is better than the window if we want to output debug information.

k-okada, 2019-03-19:
what you can see on OpenCV window should be published as ros topic.
https://github.com/ros-perception/opencv_apps/blob/indigo/src/nodelet/adding_images_nodelet.cpp#L221
and our test code also confirm that.
https://github.com/ros-perception/opencv_apps/blob/indigo/src/nodelet/adding_images_nodelet.cpp#L221
--
◉ Kei Okada

fujimo-t, 2019-03-20:
I'm sorry for spending your time.

I somehow misunderstand that there aren't ros topic for debug output.
This should be closed.

closed 2019-03-20 by fujimo-t