ros-perception/opencv_apps #90: Error about face_recognition.launch

tongtybj, 2019-04-19:
Any suggestion about the following error?
The face detection part works well.

```
nishio@nishio-desktop:~$ roslaunch opencv_apps face_recognition.launch  image:=/zed/left/image_raw_color use_opencv3:=true  use_opencv3_3:=true
... logging to /home/nishio/.ros/log/8f8e5f22-624f-11e9-989d-88d7f6d5c690/roslaunch-nishio-desktop-16328.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://nishio-desktop:41435/

SUMMARY
========

PARAMETERS
 * /face_detection/debug_view: True
 * /face_detection/eyes_cascade_name: /opt/ros/kinetic/...
 * /face_detection/face_cascade_name: /opt/ros/kinetic/...
 * /face_detection/use_camera_info: False
 * /face_recognition/data_dir: ~/.ros/opencv_app...
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    debug_image_viewer_nishio_desktop_16328_8344988699730467132 (image_view/image_view)
    face_detection (opencv_apps/face_detection)
    face_recognition (opencv_apps/face_recognition)
    face_recognition_trainer (opencv_apps/face_recognition_trainer.py)

ROS_MASTER_URI=http://localhost:11311

process[face_detection-1]: started with pid [16345]
process[face_recognition-2]: started with pid [16346]
process[face_recognition_trainer-3]: started with pid [16347]
process[debug_image_viewer_nishio_desktop_16328_8344988699730467132-4]: started with pid [16350]
[ INFO] [1555644640.109175425]: Initializing nodelet with 1 worker threads.
[ERROR] [1555644640.252733270]: --Error loading /opt/ros/kinetic/share/opencv3/../OpenCV-3.3.1/haarcascades/haarcascade_frontalface_alt.xml
[ERROR] [1555644640.252811938]: --Error loading /opt/ros/kinetic/share/opencv3/../OpenCV-3.3.1/haarcascades/haarcascade_eye_tree_eyeglasses.xml
OpenCV Error: Assertion failed (!empty()) in detectMultiScale, file /tmp/binarydeb/ros-kinetic-opencv3-3.3.1/modules/objdetect/src/cascadedetect.cpp, line 1698
[ERROR] [1555644640.504969826]: Image processing error: !empty() detectMultiScale /tmp/binarydeb/ros-kinetic-opencv3-3.3.1/modules/objdetect/src/cascadedetect.cpp 1698
OpenCV Error: Assertion failed (!empty()) in detectMultiScale, file /tmp/binarydeb/ros-kinetic-opencv3-3.3.1/modules/objdetect/src/cascadedetect.cpp, line 1698
[ERROR] [1555644640.534175926]: Image processing error: !empty() detectMultiScale /tmp/binarydeb/ros-kinetic-opencv3-3.3.1/modules/objdetect/src/cascadedetect.cpp 1698
OpenCV Error: Assertion failed (!empty()) in detectMultiScale, file /tmp/binarydeb/ros-kinetic-opencv3-3.3.1/modules/objdetect/src/cascadedetect.cpp, line 1698
[ERROR] [1555644640.564517071]: Image processing error: !empty() detectMultiScale /tmp/binarydeb/ros-kinetic-opencv3-3.3.1/modules/objdetect/src/cascadedetect.cpp 1698
OpenCV Error: Assertion failed (!empty()) in detectMultiScale, file /tmp/binarydeb/ros-kinetic-opencv3-3.3.1/modules/objdetect/src/cascadedetect.cpp, line 1698
[ERROR] [1555644640.598023871]: Image processing error: !empty() detectMultiScale /tmp/binarydeb/ros-kinetic-opencv3-3.3.1/modules/objdetect/src/cascadedetect.cpp 1698
OpenCV Error: Assertion failed (!empty()) in detectMultiScale, file /tmp/binarydeb/ros-kinetic-opencv3-3.3.1/modules/objdetect/src/cascadedetect.cpp, line 1698
[ERROR] [1555644640.631610306]: Image processing error: !empty() detectMultiScale /tmp/binarydeb/ros-kinetic-opencv3-3.3.1/modules/objdetect/src/cascadedetect.cpp 1698
OpenCV Error: Assertion failed (!empty()) in detectMultiScale, file /tmp/binarydeb/ros-kinetic-opencv3-3.3.1/modules/objdetect/src/cascadedetect.cpp, line 1698
[ERROR] [1555644640.664901007]: Image processing error: !empty() detectMultiScale /tmp/binarydeb/ros-kinetic-opencv3-3.3.1/modules/objdetect/src/cascadedetect.cpp 1698
OpenCV Error: Assertion failed (!empty()) in detectMultiScale, file /tmp/binarydeb/ros-kinetic-opencv3-3.3.1/modules/objdetect/src/cascadedetect.cpp, line 1698
[ERROR] [1555644640.699107667]: Image processing error: !empty() detectMultiScale /tmp/binarydeb/ros-kinetic-opencv3-3.3.1/modules/objdetect/src/cascadedetect.cpp 1698
OpenCV Error: Assertion failed (!empty()) in detectMultiScale, file /tmp/binarydeb/ros-kinetic-opencv3-3.3.1/modules/objdetect/src/cascadedetect.cpp, line 1698
[ERROR] [1555644640.732061251]: Image processing error: !empty() detectMultiScale /tmp/binarydeb/ros-kinetic-opencv3-3.3.1/modules/objdetect/src/cascadedetect.cpp 1698
OpenCV Error: Assertion failed (!empty()) in detectMultiScale, file /tmp/binarydeb/ros-kinetic-opencv3-3.3.1/modules/objdetect/src/cascadedetect.cpp, line 1698
[ERROR] [1555644640.767059899]: Image processing error: !empty() detectMultiScale /tmp/binarydeb/ros-kinetic-opencv3-3.3.1/modules/objdetect/src/cascadedetect.cpp 1698
OpenCV Error: Assertion failed (!empty()) in detectMultiScale, file /tmp/binarydeb/ros-kinetic-opencv3-3.3.1/modules/objdetect/src/cascadedetect.cpp, line 1698
[ERROR] [1555644640.798270680]: Image processing error: !empty() detectMultiScale /tmp/binarydeb/ros-kinetic-opencv3-3.3.1/modules/objdetect/src/cascadedetect.cpp 1698
OpenCV Error: Assertion failed (!empty()) in detectMultiScale, file /tmp/binarydeb/ros-kinetic-opencv3-3.3.1/modules/objdetect/src/cascadedetect.cpp, line 1698
[ERROR] [1555644640.831769118]: Image processing error: !empty() detectMultiScale /tmp/binarydeb/ros-kinetic-opencv3-3.3.1/modules/objdetect/src/cascadedetect.cpp 1698
OpenCV Error: Assertion failed (!empty()) in detectMultiScale, file /tmp/binarydeb/ros-kinetic-opencv3-3.3.1/modules/objdetect/src/cascadedetect.cpp, line 1698
[ERROR] [1555644640.865115907]: Image processing error: !empty() detectMultiScale /tmp/binarydeb/ros-kinetic-opencv3-3.3.1/modules/objdetect/src/cascadedetect.cpp 1698
OpenCV Error: Assertion failed (!empty()) in detectMultiScale, file /tmp/binarydeb/ros-kinetic-opencv3-3.3.1/modules/objdetect/src/cascadedetect.cpp, line 1698
[ERROR] [1555644640.897904403]: Image processing error: !empty() detectMultiScale /tmp/binarydeb/ros-kinetic-opencv3-3.3.1/modules/objdetect/src/cascadedetect.cpp 1698
OpenCV Error: Assertion failed (!empty()) in detectMultiScale, file /tmp/binarydeb/ros-kinetic-opencv3-3.3.1/modules/objdetect/src/cascadedetect.cpp, line 1698
```
[image]

k-okada, 2019-04-19:
please use
https://raw.githubusercontent.com/ros-perception/opencv_apps/f662a1b9315d463e510d54934b17cc1775296325/launch/face_detection.launch,
we will release this version very soon
--
◉ Kei Okada

knorth55, 2019-04-19:
https://github.com/ros-perception/opencv_apps/commit/f662a1b9315d463e510d54934b17cc1775296325
This commit fix this issue

tongtybj, 2019-04-19:
sorry, I **pasted** the wrong error log. The path had been fixed by myself in local pc.
The true error log  belonging to the face_recognition is as follows:

```
nishio@nishio-desktop:/opt/ros/kinetic/share/opencv_apps/launch$ roslaunch opencv_apps face_recognition.launch use_opencv3:=true  image:=/zed/left/image_raw_color use_opencv3_3:=true
... logging to /home/nishio/.ros/log/8f8e5f22-624f-11e9-989d-88d7f6d5c690/roslaunch-nishio-desktop-20379.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://nishio-desktop:38085/

SUMMARY
========

PARAMETERS
 * /face_detection/debug_view: True
 * /face_detection/eyes_cascade_name: /opt/ros/kinetic/...
 * /face_detection/face_cascade_name: /opt/ros/kinetic/...
 * /face_detection/use_camera_info: False
 * /face_recognition/data_dir: ~/.ros/opencv_app...
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    debug_image_viewer_nishio_desktop_20379_402860772321071504 (image_view/image_view)
    face_detection (opencv_apps/face_detection)
    face_recognition (opencv_apps/face_recognition)
    face_recognition_trainer (opencv_apps/face_recognition_trainer.py)

ROS_MASTER_URI=http://localhost:11311

process[face_detection-1]: started with pid [20396]
process[face_recognition-2]: started with pid [20397]
process[face_recognition_trainer-3]: started with pid [20399]
process[debug_image_viewer_nishio_desktop_20379_402860772321071504-4]: started with pid [20401]
[ INFO] [1555647572.644758471]: Initializing nodelet with 1 worker threads.
OpenCV Error: Unspecified error (This Eigenfaces model is not computed yet. Did you call Eigenfaces::train?) in predict, file /tmp/binarydeb/ros-kinetic-opencv3-3.3.1/opencv_contrib/face/src/eigen_faces.cpp, line 117
[ERROR] [1555647573.478618348]: error at image processing: This Eigenfaces model is not computed yet. Did you call Eigenfaces::train? predict /tmp/binarydeb/ros-kinetic-opencv3-3.3.1/opencv_contrib/face/src/eigen_faces.cpp 117
OpenCV Error: Unspecified error (This Eigenfaces model is not computed yet. Did you call Eigenfaces::train?) in predict, file /tmp/binarydeb/ros-kinetic-opencv3-3.3.1/opencv_contrib/face/src/eigen_faces.cpp, line 117
[ERROR] [1555647573.581168695]: error at image processing: This Eigenfaces model is not computed yet. Did you call Eigenfaces::train? predict /tmp/binarydeb/ros-kinetic-opencv3-3.3.1/opencv_contrib/face/src/eigen_faces.cpp 117
OpenCV Error: Unspecified error (This Eigenfaces model is not computed yet. Did you call Eigenfaces::train?) in predict, file /tmp/binarydeb/ros-kinetic-opencv3-3.3.1/opencv_contrib/face/src/eigen_faces.cpp, line 117
[ERROR] [1555647573.685594093]: error at image processing: This Eigenfaces model is not computed yet. Did you call Eigenfaces::train? predict /tmp/binarydeb/ros-kinetic-opencv3-3.3.1/opencv_contrib/face/src/eigen_faces.cpp 117
OpenCV Error: Unspecified error (This Eigenfaces model is not computed yet. Did you call Eigenfaces::train?) in predict, file /tmp/binarydeb/ros-kinetic-opencv3-3.3.1/opencv_contrib/face/src/eigen_faces.cpp, line 117
[ERROR] [1555647573.785534746]: error at image processing: This Eigenfaces model is not computed yet. Did you call Eigenfaces::train? predict /tmp/binarydeb/ros-kinetic-opencv3-3.3.1/opencv_contrib/face/src/eigen_faces.cpp 117
OpenCV Error: Unspecified error (This Eigenfaces model is not computed yet. Did you call Eigenfaces::train?) in predict, file /tmp/binarydeb/ros-kinetic-opencv3-3.3.1/opencv_contrib/face/src/eigen_faces.cpp, line 117
[ERROR] [1555647573.885539787]: error at image processing: This Eigenfaces model is not computed yet. Did you call Eigenfaces::train? predict /tmp/binarydeb/ros-kinetic-opencv3-3.3.1/opencv_contrib/face/src/eigen_faces.cpp 117
OpenCV Error: Unspecified error (This Eigenfaces model is not computed yet. Did you call Eigenfaces::train?) in predict, file /tmp/binarydeb/ros-kinetic-opencv3-3.3.1/opencv_contrib/face/src/eigen_faces.cpp, line 117
[ERROR] [1555647573.985925666]: error at image processing: This Eigenfaces model is not computed yet. Did you call Eigenfaces::train? predict /tmp/binarydeb/ros-kinetic-opencv3-3.3.1/opencv_contrib/face/src/eigen_faces.cpp 117
OpenCV Error: Unspecified error (This Eigenfaces model is not computed yet. Did you call Eigenfaces::train?) in predict, file /tmp/binarydeb/ros-kinetic-opencv3-3.3.1/opencv_contrib/face/src/eigen_faces.cpp, line 117
[ERROR] [1555647574.085516911]: error at image processing: This Eigenfaces model is not computed yet. Did you call Eigenfaces::train? predict /tmp/binarydeb/ros-kinetic-opencv3-3.3.1/opencv_contrib/face/src/eigen_faces.cpp 117
OpenCV Error: Unspecified error (This Eigenfaces model is not computed yet. Did you call Eigenfaces::train?) in predict, file /tmp/binarydeb/ros-kinetic-opencv3-3.3.1/opencv_contrib/face/src/eigen_faces.cpp, line 117
[ERROR] [1555647574.184622094]: error at image processing: This Eigenfaces model is not computed yet. Did you call Eigenfaces::train? predict /tmp/binarydeb/ros-kinetic-opencv3-3.3.1/opencv_contrib/face/src/eigen_faces.cpp 117
OpenCV Error: Unspecified error (This Eigenfaces model is not computed yet. Did you call Eigenfaces::train?) in predict, file /tmp/binarydeb/ros-kinetic-opencv3-3.3.1/opencv_contrib/face/src/eigen_faces.cpp, line 117
```
[image]

k-okada, 2019-04-19:
okay....

1) please do not edit files with `sudo`. you can use `sudo` only for `apt` command. if someone told you to change the code or run command by `sudo emacs` or `sudo vi`, that's is a very bad person. 
2) if you ask someone, please provide how to reproduce your problem in other's computer. for example, you can run `test-face_recognition.test` with bag file `http://download.ros.org/data/face_detector/face_detector_withface_test_diamondback.bag
` or you can provide your bag file
3) please add expected behavior and why you think you have a problem

k-okada, 2019-04-19:
please check contents of `~/.ros/opencv_apps/face_data/`, is there is no image file, you need to  take pictures of your face from face_recognition_trainer.py window

tongtybj, 2019-04-19:
@k-okada 

Thank you for the important advice and instruction.
I was the bad guy being reformed, and have checked the expected behaviour by `test-face_recognition.test.` 

Right now I know the the pasted error log is from [here](https://github.com/ros-perception/opencv_apps/blob/indigo/src/nodelet/face_recognition_nodelet.cpp#L375-L381), caused because of  no trained data. 
So it always occurs in the taking pictures phase.

Finally, the expected result is also confirmed with my face.

Suggestion:
Maybe, it is better to add a line [here](https://github.com/ros-perception/opencv_apps/blob/indigo/src/nodelet/face_recognition_nodelet.cpp#L347), to check the size of the `label_mapper_`. If it is empty, then just `return` from this callback, avoiding the confusion. 
[image]

k-okada, 2019-04-19:
> Maybe, it is better to add a line here, to check the size of the
label_mapper_. If it is empty, then just return from this callback,
avoiding the confusion.

super great idea !!! please create PR for this.
--
◉ Kei Okada

tongtybj, 2019-04-19:
Please check #91 .