ros-perception/opencv_apps #109: Histogram equalization

ivan1993br, 2020-10-08:
Hi all,
I was just wondering if a Histogram equalization would be in the scope of this package, I didn't found any ROS Node that does that, so I was thinking about adding it here.

What do you think @k-okada ?

Thanks!

k-okada, 2020-10-09:
@ivan1993br thank you for proposal Yes, addiing histgram equalization would be nice. This package is basically ROS node of opencv sample program and there already exsits https://github.com/opencv/opencv/tree/master/samples/cpp/tutorial_code/Histograms_Matching. 
I have added some information on how to contirbute. so please look at 
https://github.com/ros-perception/opencv_apps/blob/contrib/CONTRIBUTING.md

closed 2022-02-11 by k-okada

bochen87, 2022-02-24:
since this feature is now available, would be really cool if it could be released as well!

k-okada, 2022-02-25:
@bochen87 thanks,
I also added node related to histogram at  https://github.com/ros-perception/opencv_apps/pull/128 in addition to https://github.com/ros-perception/opencv_apps/pull/126. Do you have any comment/review on that?

bochen87, 2022-02-25:
> 
Yes, I tested #126 with clahe to improve our image preprocessing pipeline for better april tag detection under different lighting conditions, works very well!

I can test out #128 next week.

Would be really awesome to release a new debian package version with all these new features!